robfiras/loco-mujoco #7: How do you get the ground truth expert data for humanoid robots?

wenshuaizhao, 2023-12-07:
Hi,

Interesting work! But I didn't get the information from the paper, how do you get the ground truth data with actions for humanoid robots, such as Atlas?

Thanks

robfiras, 2023-12-08:
We optimize the motion capture towards a base humanoid model using Qualsys' optimization software (see paper for reference). Then we manually adapt this data to the different humanoid embodiment using affine transformations. To get the actions, we use the baseline algorithms provided to learn a policy that imitates the motion capture data as good as possible. We will provide the experimental setup, such that you can train an agent yourself if you want.

wenshuaizhao, 2023-12-09:
Thanks for your detailed explanation!

closed 2023-12-09 by wenshuaizhao