robfiras/loco-mujoco #8: Need details on how the '*.npz' dataset files mentioned in the project are generated.

wangshub, 2023-12-16:
Hi @robfiras , thank you very much for your excellent work !

 I am currently attempting to implement motion capture data learning on the `Kuavo` robot model. 

- https://github.com/wangshub/loco-mujoco/tree/master/loco_mujoco/environments/data/kuavo
 
After reading through your documentation and paper, I'm still confused about how files like `05-run_ATLAS.npz、02-constspeed_UnitreeH1.npz ...` are generated. 

- Could you provide details on the process behind creating these files?
- Could you provide an explanation of the format of `NPZ` files?

Looking forward to your response!

robfiras, 2023-12-30:
Hi!
sorry for the late reply..
I took a look into your model, and it looks very nice!
The datasets are collected using motion capture. We optimize the motion capture towards the base humanoid (the skeleton model) model using Qualsys' optimization software (see paper for reference). Then we manually adapt this data to the different humanoid embodiments (UnitreeH1, Atlas, Talos etc) using affine transformations. So what you would have to do is to checkout the data from the base humanoid, check which joints corresponds to joints in your humanoid model, and then fine-tune the trajectory a bit to match your robot's kinematics. 
The NPZ files contain all trajectories *per joint*. The split points indicate the start of the trajectories. 
If you want to include the kuavo robot in LocoMuJoCo, I can create the trajectory files for it. I would ask you the to create a PR to include the kuavo. Right now, the meshes used for the kuavo are somewhat basic, it would be super cool if you could make it look more appealing! 
Best wishes,
Firas

philfung, 2023-12-31:
hi @robfiras 

>  We optimize the motion capture towards the base humanoid (the skeleton model) model using Qualsys' optimization software (see paper for reference). Then we manually adapt this data to the different humanoid embodiments (UnitreeH1, Atlas, Talos etc) using affine transformations

Would you have any example code to show how you do the manually adaptation?

Thanks in any case for making such a cool library!

robfiras, 2024-01-02:
Hi @philfung 

Due to popular requests, I am working on releasing this pipeline. I will notify you once it's done (:

evgps, 2024-02-06:
> Hi @philfung
> 
> Due to popular requests, I am working on releasing this pipeline. I will notify you once it's done (:

Dear @robfiras, how do you think, when this pipeline will be ready?

robfiras, 2024-02-06:
Hi @evgps,
I will try to finish it by the end of next week (:

robfiras, 2024-02-06:
Hi @wangshub @philfung @evgps,
I actually had time today to add the data generation pipeline. You can find it [https://github.com/robfiras/loco-mujoco/tree/master/loco_mujoco/datasets/data_generation](url). It would be great if you could take a look! Just tell me if there is anything, which is not clear yet.

philfung, 2024-02-09:
hi @robfiras thanks so much for sharing this!  Initially I had thought there was a theory-based mapping of the mocap data to the model, instead it seems these mappings were made by manually adapting weights based on visual output, is that correct?  In that case, it makes a lot of sense!  Thanks again for doing such great work.

robfiras, 2024-02-09:
Hi @philfung,
yes. As mentioned before (and in the paper) the mapping from key points to the base humanoid joint angles is done using Qualsys' optimization software. However that one is proprietary. The adaption for each humanoid is then done by manually fine tuning the joint angles. We are working on an automatic optimizer taking 3d coordinates and mapping them to the joint angles of each humanoid, but this will take some time. 
I will close this issue for now, but feel free to open it again in case something is still unclear.

closed 2024-02-09 by robfiras

wangshub, 2024-02-21:
@philfung I have made the Kuavo humanoid work.. check this out  https://github.com/robfiras/loco-mujoco/pull/12

@robfiras