robfiras/loco-mujoco #11: Inference code

emedinac, 2024-02-13:
Hi,

I would like to know if you have an inference code when the model is trained using your launcher code. Is there any template to test different training?

Tranks!

robfiras, 2024-02-13:
Hi!
Yeah, it actually very simple. Here it is:

``` python
from mushroom_rl.core import Core, Agent
from loco_mujoco import LocoEnv

env = LocoEnv.make("Atlas.walk")

agent = Agent.load("./path/to/agent.msh")

core = Core(agent, env)

core.evaluate(n_episodes=10, render=True)
```

robfiras, 2024-02-13:
If you want to evaluate a lot of agents with the launcher (rather than watching a trained one), you have to modify the experiment script from the examples to load an agent, instead of creating a new one (see example above). Also, comment out the `core.lean(...)` function, as you only want to do inference. The evaluation calls are already done in the experiment file. I don't have a script at hand right now, but it should be very straight forward to do. Let me know if you need further help with this!

closed 2024-02-13 by robfiras