robfiras/loco-mujoco #16: I get an error：urllib.error.URLError: <urlopen error [Errno 111] Connection refused>

2802029572, 2024-03-05:
Hello, when I am using loco-mujoco-download to download the dataset, I get an error：urllib.error.URLError: <urlopen error [Errno 111] Connection refused>, is there any other way to download the dataset?

robfiras, 2024-03-05:
Hi, 
today, it seems like there is an issue with Zenodo's dataset hosting service (which we use for the locomujoco datasets). We can not access the dataset there right now. There is no problem with LocoMujoco. 
Here is where the datasets are: https://zenodo.org/records/10102870
Right now their download button is not working.

We will contact them and check what the problem is. Thanks for notifying us, I will get back to you.

robfiras, 2024-03-05:
They wrote back to me. They are having some issues on their server side. This will be fixed soon.

robfiras, 2024-03-05:
Works again, closing this.

closed 2024-03-05 by robfiras

2802029572, 2024-03-06:
Sorry, I downloaded the database using the website, but I don't know exactly which folder to put it in.In the meantime, I'm still having the same problem with loco-mujoco-download.

2802029572, 2024-03-06:
My mistake was:UserWarning: Datasets not found, falling back to test datasets. Please download and install the datasets to use this environment for imitation learning!
  warnings.warn("Datasets not found, falling back to test datasets. Please download and install "