robfiras/loco-mujoco #17: Error when running pip install loco-mujoco

ShadowDawg, 2024-03-06:
I am trying to run loco-mujoco on a windows 11 machine within a conda environment.
On running the command `pip install loco-mujoco` I get the following error:

```
Building wheels for collected packages: mujoco
  Building wheel for mujoco (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for mujoco (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [79 lines of output]
      running bdist_wheel
      running build
      running build_py
      creating build
      creating build\lib.win-amd64-cpython-312
      creating build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\bindings_test.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\gl_context.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\renderer.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\renderer_test.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\render_test.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\rollout.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\rollout_test.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\viewer.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\viewer_test.py -> build\lib.win-amd64-cpython-312\mujoco
      copying mujoco\__init__.py -> build\lib.win-amd64-cpython-312\mujoco
      creating build\lib.win-amd64-cpython-312\mujoco\cgl
      copying mujoco\cgl\cgl.py -> build\lib.win-amd64-cpython-312\mujoco\cgl
      copying mujoco\cgl\__init__.py -> build\lib.win-amd64-cpython-312\mujoco\cgl
      creating build\lib.win-amd64-cpython-312\mujoco\egl
      copying mujoco\egl\egl_ext.py -> build\lib.win-amd64-cpython-312\mujoco\egl
      copying mujoco\egl\__init__.py -> build\lib.win-amd64-cpython-312\mujoco\egl
      creating build\lib.win-amd64-cpython-312\mujoco\glfw
      copying mujoco\glfw\__init__.py -> build\lib.win-amd64-cpython-312\mujoco\glfw
      creating build\lib.win-amd64-cpython-312\mujoco\osmesa
      copying mujoco\osmesa\__init__.py -> build\lib.win-amd64-cpython-312\mujoco\osmesa
      running build_ext
      Traceback (most recent call last):
        File "C:\Python\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 353, in <module>
          main()
        File "C:\Python\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Python\Lib\site-packages\pip\_vendor\pyproject_hooks\_in_process\_in_process.py", line 251, in build_wheel
          return _build_backend().build_wheel(wheel_directory, config_settings,
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\setuptools\build_meta.py", line 410, in build_wheel
          return self._build_with_temp_dir(
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\setuptools\build_meta.py", line 395, in _build_with_temp_dir
          self.run_setup()
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\setuptools\build_meta.py", line 487, in run_setup
          super().run_setup(setup_script=setup_script)
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\setuptools\build_meta.py", line 311, in run_setup
          exec(code, locals())
        File "<string>", line 419, in <module>
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\setuptools\__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\setuptools\_distutils\core.py", line 185, in setup
          return run_commands(dist)
                 ^^^^^^^^^^^^^^^^^^
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\setuptools\_distutils\core.py", line 201, in run_commands
          dist.run_commands()
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\setuptools\_distutils\dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\setuptools\dist.py", line 963, in run_command
          super().run_command(command)
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\wheel\bdist_wheel.py", line 368, in run
          self.run_command("build")
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\setuptools\_distutils\cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\setuptools\dist.py", line 963, in run_command
          super().run_command(command)
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\setuptools\_distutils\command\build.py", line 131, in run
          self.run_command(cmd_name)
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\setuptools\_distutils\cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\setuptools\dist.py", line 963, in run_command
          super().run_command(command)
        File "C:\Users\shado\AppData\Local\Temp\pip-build-env-n_n5wegf\overlay\Lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "<string>", line 156, in run
        File "<string>", line 170, in _find_mujoco
      RuntimeError: MUJOCO_PATH environment variable is not set
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for mujoco
Failed to build mujoco
ERROR: Could not build wheels for mujoco, which is required to install pyproject.toml-based projects
```

Would appreciate any help on this!

robfiras, 2024-03-06:
Hi!
at the moment we require mujoco 2.3.6. The latter is not available for python 3.12. Could you use python 3.10? 3.11 should also work, but there is some known issues on Windows.

ShadowDawg, 2024-03-06:
I changed the python version and it works now. Thanks for the quick response!

closed 2024-03-06 by ShadowDawg