robfiras/loco-mujoco #18: Numpy error when installing raw motion capture data

ShadowDawg, 2024-03-06:
I was following the instructions given [here](https://github.com/robfiras/loco-mujoco/blob/master/loco_mujoco/datasets/data_generation/README.md). 
On running 
```
from loco_mujoco.utils.dataset import download_raw_mocap_datasets
download_raw_mocap_datasets()
```
the following error occurs:
```
Unexpected exception formatting exception. Falling back to standard exception
Traceback (most recent call last):
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\numpy\core\__init__.py", line 24, in <module>
    from . import multiarray
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\numpy\core\multiarray.py", line 10, in <module>
    from . import overrides
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\numpy\core\overrides.py", line 8, in <module>
    from numpy.core._multiarray_umath import (
ModuleNotFoundError: No module named 'numpy.core._multiarray_umath'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\numpy\__init__.py", line 144, in <module>
    from numpy.__config__ import show as show_config
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\numpy\__config__.py", line 4, in <module>
    from numpy.core._multiarray_umath import (
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\numpy\core\__init__.py", line 50, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy C-extensions failed. This error can happen for
many reasons, often due to issues with your setup or how NumPy was
installed.

We have compiled some common reasons and troubleshooting tips at:

    https://numpy.org/devdocs/user/troubleshooting-importerror.html

Please note and check the following:

  * The Python version is: Python3.12 from "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\python.exe"
  * The NumPy version is: "1.26.4"

and make sure that they are the versions you expect.
Please carefully study the documentation linked above for further help.

Original error was: No module named 'numpy.core._multiarray_umath'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\IPython\core\interactiveshell.py", line 3577, in run_code
  File "C:\Users\shado\AppData\Local\Temp\ipykernel_6976\1390309027.py", line 1, in <module>
    from loco_mujoco.utils.dataset import download_raw_mocap_datasets
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\loco_mujoco\utils\__init__.py", line 1, in <module>
    from .reward import *
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\loco_mujoco\utils\reward.py", line 1, in <module>
    import numpy as np
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\numpy\__init__.py", line 149, in <module>
    raise ImportError(msg) from e
ImportError: Error importing numpy: you should not try to import numpy from
        its source directory; please exit the numpy source tree, and relaunch
        your python interpreter from there.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\pygments\styles\__init__.py", line 45, in get_style_by_name
ModuleNotFoundError: No module named 'pygments.styles.default'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\IPython\core\interactiveshell.py", line 2168, in showtraceback
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\IPython\core\ultratb.py", line 1454, in structured_traceback
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\IPython\core\ultratb.py", line 1345, in structured_traceback
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\IPython\core\ultratb.py", line 1192, in structured_traceback
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\IPython\core\ultratb.py", line 1082, in format_exception_as_a_whole
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\IPython\core\ultratb.py", line 1133, in get_records
  File "c:\Users\shado\OneDrive\Desktop\work\ai-humanoid\.conda\Lib\site-packages\pygments\styles\__init__.py", line 47, in get_style_by_name
pygments.util.ClassNotFound: Could not find style module 'pygments.styles.default', though it should be builtin.
```
My system details -
OS: Windows 11
Python version: 3.10.13 (conda)
numpy version: 1.26.4

I would appreciate any help regarding this!

robfiras, 2024-03-18:
Could you try to run this from plain python rather the IPython?

ShadowDawg, 2024-04-11:
Yes, it works from python. Thanks for the help and sorry for the late response!

closed 2024-04-11 by ShadowDawg