robfiras/loco-mujoco #20: 'dict' object has no attribute 'env_specs'

nikisim, 2024-03-22:
Hi!

I am trying to run: ```python3 examples/simple_mushroom_env/example_unitree_a1.py```
Also I get the same error when I run ```loco-mujoco-download```

But I get an error: 
```
File /loco-mujoco/loco_mujoco/environments/base.py:12
    if id in registry.env_specs:
AttributeError: 'dict' object has no attribute 'env_specs'
```

I guess it's something about package versions...

I would appreciate any help.

Ubuntu: 22.04
Python: 3.10.12

robfiras, 2024-03-22:
Hi!
Is this something specific to the A1 environment or does it happen for all environments?
Did you install the package from pip or did you do an editable installation from github?

robfiras, 2024-03-22:
Could you send the whole traceback of the error?

nikisim, 2024-03-22:
Yeah, sure!

```
Traceback (most recent call last):
  File "/home/nikisim/user_folder/loco-mujoco/examples/simple_mushroom_env/example_humanoid_muscle.py", line 2, in <module>
    from loco_mujoco import LocoEnv
  File "/home/nikisim/user_folder/loco-mujoco/loco_mujoco/__init__.py", line 5, in <module>
    from .environments import LocoEnv
  File "/home/nikisim/user_folder/loco-mujoco/loco_mujoco/environments/__init__.py", line 1, in <module>
    from .base import LocoEnv, ValidTaskConf
  File "/home/nikisim/user_folder/loco-mujoco/loco_mujoco/environments/base.py", line 12, in <module>
    from mushroom_rl.core import Environment
  File "/home/nikisim/.local/lib/python3.10/site-packages/mushroom_rl/core/__init__.py", line 7, in <module>
    import mushroom_rl.environments
  File "/home/nikisim/.local/lib/python3.10/site-packages/mushroom_rl/environments/__init__.py", line 10, in <module>
    from .gym_env import Gym
  File "/home/nikisim/.local/lib/python3.10/site-packages/mushroom_rl/environments/gym_env.py", line 9, in <module>
    import pybullet_envs
  File "/home/nikisim/.local/lib/python3.10/site-packages/pybullet_envs/__init__.py", line 14, in <module>
    register(
  File "/home/nikisim/.local/lib/python3.10/site-packages/pybullet_envs/__init__.py", line 6, in register
    if id in registry.env_specs:
AttributeError: 'dict' object has no attribute 'env_specs'
```

robfiras, 2024-03-22:
The problem seems to be Gym, which is not used within LocoMuJoCo (only Gymnasium). Mushroom-Rl requires a specific version of Gym. Using another version causes this issue. Could you try to uninstall it?

nikisim, 2024-03-22:
Yes, it works now.
Thanks a lot!

By the way, have you thought about datasets compatibility with Minari? 
I want to use Offline RL algorithms such as IQL and ReBRAC with dataset from Unitree A1 (using simulation and real robot) in my study.

robfiras, 2024-03-22:
Great!

Thanks for pointing this out. This could be an idea for the future, even though this repo focuses on imitation learning rather than offline rl.

But if I am not wrong, you could use Minari's Datacollector to create datasets in their format (if you need it). It is just a wrapper around a Gymnasium environment, which we provide.

closed 2024-03-22 by robfiras

nikisim, 2024-03-22:
> you could use Minari's Datacollector to create datasets in their format 

Yep, that's the plan :)

Anyway, thanks again for your super-fast help!

robfiras, 2024-03-22:
You're welcome!

Fyi, if you are planning to use custom rewards with Unitree A1 environment, use an editable installation of LocoMuJoCo for now. There was a bug #15, which was recently fixed. A new version will be released in the next few days.

nikisim, 2024-03-25:
Hello again...
Sorry for bothering you, but I'm encountering difficulties in obtaining a dataset for the Unitree A1 in RL format (such d4rl datasets).

I am expecting to get dataset as dict with 'observations', 'actions', 'rewards' for Unitree A1 env. But i can see that default dataset for Unitree A1 contains only state and next state.

Is it possible for me to acquire a dataset that includes the actions for walking straight? Or perhaps you possess one already?

P.S. To use Minari I still must have actions from some pretrained algorithm, but now I abandoned the idea with Minari

robfiras, 2024-03-25:
Hi,

LocoMujoco has different types of datasets, two of which are `real` and `perfect`. The `real` datasets are collected via motion capture*. These datasets come with additional challenges like not having access to the actions or having some correspondence mismatch. The `perfect` dataset are datasets collected within LocoMujoco using the imitation learning baseline we have for the respective `real` datasets. These datasets come with actions and have no correspondence mismatch. 

It is super easy to change the dataset type.

**Real:**
```python
mdp = LocoEnv.make("UnitreeA1.simple.real")
```

**Perfect:**
```python
mdp = LocoEnv.make("UnitreeA1.simple.perfect")
```
If do not specify the dataset type, `real` is chosen by default. 

*True for all environments except the UnitreeA1, where we used a MPC controller instead

nikisim, 2024-03-25:
:man_facepalming: 
Yeah, I forgot about different types of datasets.

Okay, so, to get a dict dataset with actions I should use ```mpd.create_dataset()```?
But it also returns a dataset only with states and next_states...

robfiras, 2024-03-25:
Oh, there seems to be a small bug in the UnitreeA1 environment then. 
For the other environments, it does return actions. Thanks for pointing this out, I will fix this now.

commit 6a9c4274 (2024-03-25): Fix UnitreeA1 dataset loader.

- now actions are also parsed when choosing the perfect dataset type.
- fixes #20.