robfiras/loco-mujoco #21: can not download the data

AlexZhangUPUPUP, 2024-03-31:
When I try to download the data, the following bug always appears：

`python`
(embodied-data-loco-mujoco) ubuntu@ubuntu-W480-VISION-W:~/qyy/humanoid-gym/humanoid$ loco-mujoco-download-real
/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/site-packages/loco_mujoco/datasets

Downloading Humanoid Datasets ...

Traceback (most recent call last):
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/urllib/request.py", line 1354, in do_open
    h.request(req.get_method(), req.selector, req.data, headers,
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/http/client.py", line 1256, in request
    self._send_request(method, url, body, headers, encode_chunked)
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/http/client.py", line 1302, in _send_request
    self.endheaders(body, encode_chunked=encode_chunked)
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/http/client.py", line 1251, in endheaders
    self._send_output(message_body, encode_chunked=encode_chunked)
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/http/client.py", line 1011, in _send_output
    self.send(msg)
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/http/client.py", line 951, in send
    self.connect()
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/http/client.py", line 1418, in connect
    super().connect()
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/http/client.py", line 922, in connect
    self.sock = self._create_connection(
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/socket.py", line 808, in create_connection
    raise err
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/socket.py", line 796, in create_connection
    sock.connect(sa)
ConnectionRefusedError: [Errno 111] Connection refused

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/bin/loco-mujoco-download-real", line 8, in <module>
    sys.exit(download_real_datasets())
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/site-packages/loco_mujoco/utils/dataset.py", line 31, in download_real_datasets
    wget.download(humanoid_url, out=dataset_path_humanoid_str)
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/site-packages/wget.py", line 526, in download
    (tmpfile, headers) = ulib.urlretrieve(binurl, tmpfile, callback)
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/urllib/request.py", line 247, in urlretrieve
    with contextlib.closing(urlopen(url, data)) as fp:
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/urllib/request.py", line 222, in urlopen
    return opener.open(url, data, timeout)
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/urllib/request.py", line 525, in open
    response = self._open(req, data)
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/urllib/request.py", line 542, in _open
    result = self._call_chain(self.handle_open, protocol, protocol +
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/urllib/request.py", line 502, in _call_chain
    result = func(*args)
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/urllib/request.py", line 1397, in https_open
    return self.do_open(http.client.HTTPSConnection, req,
  File "/home/ubuntu/.conda/envs/embodied-data-loco-mujoco/lib/python3.8/urllib/request.py", line 1357, in do_open
    raise URLError(err)
urllib.error.URLError: <urlopen error [Errno 111] Connection refused>

Can you help me to fix the bug? I'm pretty sure my newwork has no problem.
Another way, can you give me a link or email so I can download data directly?

``

robfiras, 2024-04-02:
are you able to download the datasets directly from here:
https://zenodo.org/records/10102870

AlexZhangUPUPUP, 2024-04-08:
Thanks a lot

closed 2024-04-08 by robfiras