robfiras/loco-mujoco #22: the motion trajectory generated by opensim into mujoco for simulation

2802029572, 2024-04-23:
Hi, do you know how to import the motion trajectory generated by opensim into mujoco for simulation? I would be grateful if you could go into details!

robfiras, 2024-04-23:
What you need is either a matlab or a numpy dataset. Both need to contain trajectory information for each joint. I would suggest you to take a look into the [data generation](https://github.com/robfiras/loco-mujoco/tree/master/loco_mujoco/datasets/data_generation) documentation, and also the format of the matlab or numpy dataset. If you decide to go for the matlab dataset, you have to follow the data generation documentation to convert it to numpy before importing in locomujoco. Let me know if you need further help with this!

2802029572, 2024-04-25:
Thank you for your reply, but I am having problems downloading the original dataset for motion capture, is it convenient to provide the initial dataset, thanks!

robfiras, 2024-04-29:
You mean you can't download them? Can you try to download them from [here](https://zenodo.org/records/10912641)?

2802029572, 2024-05-08:
Thank you for your reply，I'm using
 from loco_mujoco.utils.dataset import download_raw_mocap_datasets
download_raw_mocap_datasets()
I'm having problems downloading the raw motion capture data, can I have the raw .mot files please?

2802029572, 2024-05-28:
Hello, thank you very much for the data support, which provides and its important help to my study. If now I want to use my own human skeleton model for data replay, how should I create my own human skeleton model in mujoco? Thank you for your help!

robfiras, 2024-05-28:
You will have to convert you models to mujoco models. To do so, you could use this [converter](https://github.com/MyoHub/myoconverter). Then you create a new environment similar to the ones already created in this repo. 

I am closing this, since it is not an issue.

closed 2024-05-28 by robfiras