robfiras/loco-mujoco #27: data help!

zhangqingbo6, 2024-05-23:
I want to use the raw data to generate trajectory data for our own robots, but I see in the data generation file that each robot has a dict which contains tuples of multipliers and offsets for each joint. I would like to take the robot H1 as an example and want to know how you determine the values in the dict. Thank you!

robfiras, 2024-05-23:
Hi!
[here](https://github.com/robfiras/loco-mujoco/tree/master/loco_mujoco/datasets/data_generation) we documented how to generate the dataset for each robot. You can check it out and do the same for yours!

zhangqingbo6, 2024-05-24:
What I want to ask is how you map the robot to the numerical value, that is, what is the basis for setting the offset and multiplier of the tuple, thank you!

zhangqingbo6, 2024-05-29:
Hello! Our robot also wants to join the ranks of these robots and make some contributions to the open source library. The actual picture of the robot is as follows. Looking forward to your reply.
[image]

robfiras, 2024-05-29:
Hi  @zhangqingbo6,
sorry for the late reply ..
Your robot looks super cool and is most welcome in LocoMuJoCo!
As of now, the multipliers and offset have to be set manually (we are working on an automated retargeting script). The signs of the multipliers are most important. An easy way to identify the signs of the multipliers is to open two Mujoco simulators, and paste the Xml of your robot in one, and the XML of the HumanoidTorque model in the other (the latter has the same signs as the dataset). Then go joint by joint and check whether they move in the same direction when changing the joint slider or not. If yes, the multiplier can be set to 1, if not set it to -1. Once you have done that, the motion should already look reasonable. The last thing to do is to set the vertical offset of your robot, and maybe adapt the multipliers for the pelvis_x and pelvis_z to reduce sliding feet. But in that stage, I can help fine tuning as well. 

Also, your XML has to fulfil some properties to work well with LocoMuJoCo:
 - it needs to be parseable with mjcf (meta elements in the xml like "frame"  are not supported there)
- we currently do not support free joints, you have to add 6 distinct (3 translation; 3 rot) joints for the main body.
- we group geoms to group=0 for visual geoms and group=1 for collision geoms (can be a primitive shape)
- actuators are expected to be limited (used for action space normalization)
- add a license on the top of your xml

If there is anything I could help you with, let me know!

zhangqingbo6, 2024-05-30:
Thank you very much for your reply! We are also very happy to contribute to loco-mujoco and add our robot! We will send you the robot's xml file after completing it later. Is that ok with you? Thank you