robfiras/loco-mujoco #29: mujoco.FatalError: OpenGL version 1.5 or higher required

PROPHETE-pro, 2024-06-03:
Dear sir,

Thank you all for contributing to the community for this repo. I ran into some problems that made me unable to run the repo, so I am asking for your kind help.

After installing the repo following all the instructions provided in the ReadME, I tried to run the command:

python examples/simple_mushroom_env/example_unitree_a1.py

However, the code ended with "mujoco.FatalError: OpenGL version 1.5 or higher required".

I have searched about the OpenGL online and tried to install it using 'sudo apt-get install libglu1-mesa-dev freeglut3-dev mesa-common-dev', but it did not help.

Do you mind if you can provide some advice about how to solve this problem? Thank you so much for helping!

robfiras, 2024-06-03:
Hi,
what os are you running? Could you try to download the latest [mujoco-binaries](https://github.com/google-deepmind/mujoco/releases) and check if you can run the `simulate` file found in `mujoco-x.x.x/bin`?

PROPHETE-pro, 2024-06-05:
> Hi, what os are you running? Could you try to download the latest [mujoco-binaries](https://github.com/google-deepmind/mujoco/releases) and check if you can run the `simulate` file found in `mujoco-x.x.x/bin`?

I am running on Ubuntu 22.04.2 LTS. I downloaded mujuco-3.1.6 and the 'simulate' file is working, but when running loco-mujoco, the error is still there.

robfiras, 2024-06-05:
So the `simulate` file can open a window and properly render everything? Maybe the issue is with `glfw`. Could you check the version? You could try to upgrade by `pip install --upgrade glfw`. It's hard to help here since I can not reproduce the error. I just found a related issue on openai/mujoco-py#378. You probably have seen them already though.

henbucuoshanghai, 2025-04-02:
> I am running on Ubuntu 22.04.2 LTS. I downloaded mujuco-3.1.6 and the 'simulate' file is working, but when running loco-mujoco, the error is still there.

same error 。mojoco runs ok,but  cocoo-py is the error