robfiras/loco-mujoco #35: Cannot run imitation learning experiment

SidPad, 2024-07-04:
Hi,

I get an error when I run the launcher.py file.
I reinstalled the experiment-launcher package, but I still could not fix this problem.

Error Message:
```
Traceback (most recent call last):
  File "/home/spyd66/loco-mujoco/loco-mujoco/examples/imitation_learning/launcher.py", line 42, in <module>
    launcher.run(LOCAL, TEST)
  File "/home/spyd66/anaconda3/lib/python3.11/site-packages/experiment_launcher/launcher.py", line 101, in run
    self._run_local(test)
  File "/home/spyd66/anaconda3/lib/python3.11/site-packages/experiment_launcher/launcher.py", line 237, in _run_local
    module = import_module(self._exp_file)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/spyd66/anaconda3/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/spyd66/loco-mujoco/loco-mujoco/examples/imitation_learning/experiment.py", line 10, in <module>
    from imitation_lib.utils import BestAgentSaver
ModuleNotFoundError: No module named 'imitation_lib
```

Your help would be greatly appreciated!

robfiras, 2024-07-04:
It looks like you did not install the imitation learning library.

SidPad, 2024-07-05:
Sorry, I should have checked properly. I thought the imitation library used is 'imitation'.
I installed the ls-iq package and now it seems to be working fine, thanks again!

closed 2024-07-05 by SidPad