robfiras/loco-mujoco #37: Extract .pt file from the .msh file

Harry-maximum, 2024-07-09:
Hi! Recently I trained my own agent in a new robot env, but I find that it uses the mushroom rl so that it save the file format of .msh, so is there any methods I can change the .msh to .pt file?

robfiras, 2024-07-11:
Hi!
after you load an agent with

```python
from mushroom_rl.core import Agent
path = "your_path"
agent  = Agent.load(path)
```
You can access the Gaussian policy's mean network with "agent.policy._mu.network" and save it however you like.

Harry-maximum, 2024-07-12:
Dear Firas,

Thank you for your reply! I can now load the .msh file to my a .pt file to
the environment.
I am working on the GR1T1 and GR1T2 from Fourier Intelligence, the expert
trajectory seems very well and behave quite like a human, but after
epochs of training, the robot can only walk using small steps. So can you
please judge my model and speculate the reasons? The training Hyper_params
just keeps similar with Unitree H1 and G1.

Best regards,
Xuanbo

 run_expert.mp4
<https://drive.google.com/file/d/1fZQ9lgWznhHIHovBI6wLWWiKHdieTxVl/view?usp=drive_web>

 walk_expert.mp4
<https://drive.google.com/file/d/1lB7-GKh90Jc-f0qitpwoOX3Fo1pW_XsA/view?usp=drive_web>

 2024-07-12 09-59-18.mp4
<https://drive.google.com/file/d/1D4ehH0u7WVGTSrY-u1hJEFd3LO2dHCR9/view?usp=drive_web>