robfiras/loco-mujoco #42: unitreea1_simple perfect datasets downloaded in wrong folder using `loco-mujoco-download`

OliEfr, 2024-08-09:
I noticed this on multiple different machines and fresh installs of loco-mujoco. I looked into the download-code briefly, but couldn't spot the bug so far.

The datasets are not downloaded in

`loco-mujoco/loco_mujoco/datasets/quadrupeds/perfect/mini_datasets/unitreea1_simple/`

but in

`loco-mujoco/loco_mujoco/datasets/quadrupeds/perfect/perfect/unitreea1_simple/perfect_expert_dataset_det.npz`

So there is one `perfect` too much.

This bug lead to an error upon environment creation. When moving the downloaded datasets to the correct path everything works as expected.

robfiras, 2024-08-09:
Hi @OliEfr 
thanks for spotting this. I found the error. It comes from the downloaded datasets. I will fix this in the upcomming version.
Your first dataset path is not correct though, it should be:
`loco-mujoco/loco_mujoco/datasets/quadrupeds/perfect/unitreea1_simple`

OliEfr, 2024-08-09:
Thanks for fixing.

Indeed, your path is the correct one.

Best,

Oli

commit 3f2ba108 (2024-09-09): Fixed minor issues for new release.

- fixed wrong paths for dataset download.
- fixed dm_control version to 1.0.14 to work with mujoco==2.3.7.
- updated readme.
- removed MyoSuiteHumanoid from test since it requires mujoco>=3.2.2.
- fixed #42 with new datasets.
closed 2024-09-09 by robfiras