robfiras/loco-mujoco #46: Question: desired velocity for UnitreeA1 as sine-cosine feature

OliEfr, 2024-08-21:
I was just wondering if there is a rational behind using the target velocity for this environment as a sine-cosine feature as input in the policy instead of the standard way of using the vector?

robfiras, 2024-09-09:
Hi @OliEfr,
sorry for the delay in response...
There is no specific reason behind it, but you can do the same thing using a vector. In fact, in the next big release, this will change. 
Since you are very active here, consider joining the discord channel I created for LocoMujoco. It is a better place to ask such questions.
Thanks for your contributions!
I just merged them.

closed 2024-09-09 by robfiras

OliEfr, 2024-09-09:
Hi @robfiras
No problem, I was just curious if there is something I didn't know about and the sin-cos features improves learning. Ok, I'll join the discord then.