robfiras/loco-mujoco #66: change MjxUnitreeH1 to MjxSkeletonMuscle,

YHFone, 2025-05-07:
Hello, thank you for your open source. Use jax_amp in the example to change MjxUnitreeH1 to MjxSkeletonMuscle, report assert self.obs_container == self.th.traj.obs_container, \
AssertionError: Observation containers of trajectory and environment do not match.

JakeFishcode, 2025-07-14:
Have you solved this problem? Can you share your experience in solving it?

Alternatively, if you want a more casual tone:
Did you figure it out? Mind sharing how you solved it?

Let me know if you'd like any adjustments!

robfiras, 2025-07-15:
Did you delete the expert_traj.npz file generated when starting the UnitreeH1 job?

YHFone, 2025-07-16:
I solved this problem in May, but forgot to update it. It is indeed only necessary to delete the expert_traj.npz file. Are you interested in artificial muscle simulation? Welcome to work in China.