robfiras/loco-mujoco #68: Training fails when switching env_name from MjxUnitreeH1 to MjxUnitreeG1 in jax_rl_mimic example

shenbiachao, 2025-05-11:
In the file loco-mujoco/examples/training_examples/jax_rl_mimic/experiment.py, the provided training setup successfully enables H1 to mimic a reference walking trajectory. 

However, when I try to adapt the same setup for G1 by modifying the env_name field in loco-mujoco/examples/training_examples/jax_rl_mimic/conf.yaml from MjxUnitreeH1 to MjxUnitreeG1, the training fails.

Could you please provide some guidance on how to properly train G1 to mimic a trajectory, or what additional modifications might be necessary to make it work?

Thank you!

rlee3359, 2025-05-17:
Hi, I also noticed this problem. When it finishes running you can see the reward has improved but is much lower than the H1. If you try training it for much longer the reward does improve but I didn't yet get it to the same performance as the H1 demo. I think there's likely some optimization to be done to tune this, perhaps in the reward or PPO parameters.

robagar, 2025-06-06:
Same result here I think - setting `env_name` to _MjxUnitreeG1_ with no other changes resulted in very chaotic robot movement, not even close to walking