robfiras/loco-mujoco #72: Training example jax_rl_mimic error - cuSolver internal error

robagar, 2025-05-22:
The `jax_rl_mimic` training example fails with this error:

```
Traceback (most recent call last):
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/train/jax_rl_mimic/experiment.py", line 37, in experiment
    env = factory.make(**config.experiment.env_params, **config.experiment.task_factory.params)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/loco_mujoco/task_factories/imitation_factory.py", line 69, in make
    env = env_cls(init_state_type=init_state_type, terminal_state_type=terminal_state_type, **kwargs)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/loco_mujoco/environments/humanoids/unitreeH1_mjx.py", line 17, in __init__
    super().__init__(timestep=timestep, n_substeps=n_substeps, model_option_conf=model_option_conf, **kwargs)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/loco_mujoco/environments/humanoids/unitreeH1.py", line 214, in __init__
    super().__init__(spec=spec, actuation_spec=actuation_spec, observation_spec=observation_spec, **kwargs)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/loco_mujoco/environments/base.py", line 77, in __init__
    super(LocoEnv, self).__init__(**core_params)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/loco_mujoco/core/mujoco_mjx.py", line 74, in __init__
    self._first_data = mjx.forward(self.sys, data)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/mujoco/mjx/_src/forward.py", line 57, in wrapper
    res = fn(*args, **kwargs)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/mujoco/mjx/_src/forward.py", line 404, in forward
    d = fwd_position(m, d)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/mujoco/mjx/_src/forward.py", line 57, in wrapper
    res = fn(*args, **kwargs)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/mujoco/mjx/_src/forward.py", line 72, in fwd_position
    d = smooth.factor_m(m, d)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/mujoco/mjx/_src/smooth.py", line 305, in factor_m
    qh, _ = jax.scipy.linalg.cho_factor(d.qM)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/jax/_src/scipy/linalg.py", line 154, in cho_factor
    return (cholesky(a, lower=lower), lower)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/jax/_src/scipy/linalg.py", line 106, in cholesky
    return _cholesky(a, lower)
jaxlib._jax.XlaRuntimeError: INTERNAL: cuSolver internal error
```

The runtime environment:
- Ubuntu 22.04
- 24GB NVIDIA GeForce RTX 4090

```
jax                      0.6.1
jax-cuda12-pjrt          0.6.1
jax-cuda12-plugin        0.6.1

nvidia-cublas-cu12       12.8.4.1
nvidia-cuda-cupti-cu12   12.9.19
nvidia-cuda-nvcc-cu12    12.9.41
nvidia-cuda-runtime-cu12 12.9.37
nvidia-cudnn-cu12        9.10.1.4
nvidia-cufft-cu12        11.4.0.6
nvidia-cusolver-cu12     11.7.4.40
nvidia-cusparse-cu12     12.5.9.5
nvidia-nccl-cu12         2.26.5
nvidia-nvjitlink-cu12    12.9.41
nvidia-nvshmem-cu12      3.2.5
```

robagar, 2025-05-22:
`jax_gail` fails too with the same error.

robagar, 2025-05-22:
Related - https://github.com/jax-ml/jax/issues/8916

robagar, 2025-05-22:
Minimal repro case - https://github.com/robagar/jax_cholesky_internal_error

robagar, 2025-05-22:
More informative error with `JAX_TRACEBACK_FILTERING=off`
```
Error executing job with overrides: []
Traceback (most recent call last):
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/train/jax_gail/experiment.py", line 40, in experiment
    env = factory.make(**config.experiment.env_params, **config.experiment.task_factory.params)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/loco_mujoco/task_factories/imitation_factory.py", line 69, in make
    env = env_cls(init_state_type=init_state_type, terminal_state_type=terminal_state_type, **kwargs)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/loco_mujoco/environments/humanoids/unitreeH1_mjx.py", line 17, in __init__
    super().__init__(timestep=timestep, n_substeps=n_substeps, model_option_conf=model_option_conf, **kwargs)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/loco_mujoco/environments/humanoids/unitreeH1.py", line 214, in __init__
    super().__init__(spec=spec, actuation_spec=actuation_spec, observation_spec=observation_spec, **kwargs)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/loco_mujoco/environments/base.py", line 77, in __init__
    super(LocoEnv, self).__init__(**core_params)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/loco_mujoco/core/mujoco_mjx.py", line 74, in __init__
    self._first_data = mjx.forward(self.sys, data)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/mujoco/mjx/_src/forward.py", line 57, in wrapper
    res = fn(*args, **kwargs)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/mujoco/mjx/_src/forward.py", line 404, in forward
    d = fwd_position(m, d)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/mujoco/mjx/_src/forward.py", line 57, in wrapper
    res = fn(*args, **kwargs)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/mujoco/mjx/_src/forward.py", line 72, in fwd_position
    d = smooth.factor_m(m, d)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/mujoco/mjx/_src/smooth.py", line 305, in factor_m
    qh, _ = jax.scipy.linalg.cho_factor(d.qM)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/jax/_src/scipy/linalg.py", line 154, in cho_factor
    return (cholesky(a, lower=lower), lower)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/jax/_src/scipy/linalg.py", line 106, in cholesky
    return _cholesky(a, lower)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/jax/_src/traceback_util.py", line 182, in reraise_with_filtered_traceback
    return fun(*args, **kwargs)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/jax/_src/pjit.py", line 334, in cache_miss
    executable, pgle_profiler) = _python_pjit_helper(fun, jit_info, *args, **kwargs)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/jax/_src/pjit.py", line 195, in _python_pjit_helper
    out_flat, compiled, profiler = _pjit_call_impl_python(*args_flat, **p.params)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/jax/_src/pjit.py", line 1862, in _pjit_call_impl_python
    ).compile()
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/jax/_src/interpreters/pxla.py", line 2467, in compile
    executable = UnloadedMeshExecutable.from_hlo(
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/jax/_src/interpreters/pxla.py", line 3009, in from_hlo
    xla_executable = _cached_compilation(
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/jax/_src/interpreters/pxla.py", line 2800, in _cached_compilation
    xla_executable = compiler.compile_or_get_cached(
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/jax/_src/compiler.py", line 447, in compile_or_get_cached
    return _compile_and_write_cache(
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/jax/_src/compiler.py", line 719, in _compile_and_write_cache
    executable = backend_compile(
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/jax/_src/profiler.py", line 354, in wrapper
    return func(*args, **kwargs)
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/jax/_src/compiler.py", line 342, in backend_compile
    raise e
  File "/home/rob/repos/rd/training-from-mocap/try_locomujoco/.venv/lib/python3.10/site-packages/jax/_src/compiler.py", line 335, in backend_compile
    return backend.compile(
jaxlib._jax.XlaRuntimeError: INTERNAL: cuSolver internal error
```

robfiras, 2025-05-27:
Had a similar issue on a newer machine with a newer cuda version and driver. Could you create a requirements.txt with these versions:
```
jax==0.6.0
jaxlib==0.6.0
ml-dtypes==0.5.1
numpy==1.26.4
opt-einsum==3.4.0
scipy==1.15.2
jax-cuda12-plugin==0.6.0
jax-cuda12-pjrt==0.6.0
nvidia-cublas-cu12==12.6.4.1
nvidia-cuda-cupti-cu12==12.6.80
nvidia-cuda-nvcc-cu12==12.9.41
nvidia-cuda-runtime-cu12==12.6.77
nvidia-cudnn-cu12==9.10.1.4
nvidia-cufft-cu12==11.3.0.4
nvidia-cusolver-cu12==11.7.1.2
nvidia-cusparse-cu12==12.5.4.2
nvidia-nccl-cu12==2.26.2
nvidia-nvjitlink-cu12==12.6.85
```

and then run:
```
pip install -r requirements.txt
```

instead of:
```
pip install jax["cuda12"]
```
Not sure what the issue is exactly, but this is rather a jax than a loco-mujoco problem.

Haoqingtianxia1997, 2025-05-29:
If you have completed the installation tutorial,  please try   pip install nvidia-cublas-cu12==12.9.0.13   after uninstall it.

robagar, 2025-06-02:
OK, so training runs with this workaround -  first normal installation with `uv add jax[cuda]`, then update the CUDA packages with `uv pip install -r requirements.txt` containing
```
jax==0.6.1
jax-cuda12-pjrt==0.6.1
jax-cuda12-plugin==0.6.1
jaxlib==0.6.1
ml-dtypes==0.5.1
nvidia-cublas-cu12>=12.9
nvidia-cuda-cupti-cu12>=12.6.80
nvidia-cuda-nvcc-cu12>=12.9.41
nvidia-cuda-runtime-cu12>=12.6.77
nvidia-cudnn-cu12>=9.10.1.4
nvidia-cufft-cu12>=11.3.0.4
nvidia-cusolver-cu12>=11.7.1.2
nvidia-cusparse-cu12>=12.5.4.2
nvidia-nccl-cu12>=2.26.2
nvidia-nvjitlink-cu12>=12.6.85
```
resulting in 
```
jax                      0.6.1
jax-cuda12-pjrt          0.6.1
jax-cuda12-plugin        0.6.1
jaxlib                   0.6.1
jaxopt                   0.8.5
jaxtyping                0.3.2
...
loco-mujoco              1.0.1
ml-dtypes                0.5.1
...
mujoco                   3.2.7
mujoco-mjx               3.2.7
numpy                    1.26.4
...
nvidia-cublas-cu12       12.9.0.13
nvidia-cuda-cupti-cu12   12.6.80
nvidia-cuda-nvcc-cu12    12.9.41
nvidia-cuda-runtime-cu12 12.6.77
nvidia-cudnn-cu12        9.10.1.4
nvidia-cufft-cu12        11.3.0.4
nvidia-cusolver-cu12     11.7.1.2
nvidia-cusparse-cu12     12.5.4.2
nvidia-nccl-cu12         2.26.2
nvidia-nvjitlink-cu12    12.6.85
nvidia-nvshmem-cu12      3.2.5

```

robagar, 2025-06-06:
Closing here -  cause is a bogus upper limit on `nvidia-cublas-cu12` in Jax

closed 2025-06-06 by robagar