robfiras/loco-mujoco #78: H1 robot ends episode immediately with zero return in Deepmimic training

ShiweiLiu2002, 2025-07-21:
Hi, thank you for this great repo!

I’ve been trying to run imitation learning using `training_examples/jax_rl_mimic/experiment.py` with the default configuration. I did not encounter any errors during training, and everything seems to run normally.

However, after training for 300 million steps (`3e8 total_timesteps`), the saved rollout video shows that the H1 robot just spasms slightly in place and does not perform any meaningful locomotion. 

https://github.com/user-attachments/assets/fa2dc388-585c-40c5-b2ce-063525698b95

Additionally, I noticed that:

* **Mean Episode Return stays at 0 throughout training**
* **Mean Episode Length is constantly 1**
[image]

So it seems like every episode terminates immediately after the first step, possibly without any significant action taken.

I didn’t change much in the default config—only modified the control strategy from torque-based to PD control using the provided `position_control` settings.

Here’s the `config.yaml` I used (very close to the example):

```yaml
defaults:
  - override hydra/job_logging: default
  - override hydra/launcher: basic

#hydra:
#  mode: MULTIRUN
#  job_logging: {}
#  hydra_logging: {}
#  sweeper:
#    params:
#      experiment.learnable_std: true, false
#      experiment.init_std: 0.1, 0.2
#      experiment.num_steps: 10, 20
#      experiment.update_epochs: 4
#      experiment.clip_eps: 0.2, 0.05
#      # env_params
#      experiment.env_params.env_name: MjxUnitreeH1.walk, MjxUnitreeH1.run

wandb:
  project: "deepmimic"

experiment:
  task_factory:
    name: ImitationFactory
    params:
      lafan1_dataset_conf:
        dataset_name:
          - walk1_subject5
  env_params:
    env_name: MjxUnitreeH1
    headless: True
    disable_arms: false
    horizon: 1200
    goal_type: GoalTrajMimic
    goal_params:
      visualize_goal: true
    control_type: ${control_config.position_control.control_type}
    control_params: ${control_config.position_control.control_params}
    reward_type: MimicReward
    reward_params:
      qpos_w_sum: 0.4
      qvel_w_sum: 0.2
      rpos_w_sum: 0.5
      rquat_w_sum: 0.3
      rvel_w_sum: 0.1
      sites_for_mimic:
        - upper_body_mimic
        - left_hand_mimic
        - left_foot_mimic
        - right_hand_mimic
        - right_foot_mimic
  hidden_layers: [512, 256]
  lr: 1e-4
  num_envs: 2048
  num_steps: 200
  total_timesteps: 3e8
  update_epochs: 4
  proportion_env_reward: 0.0
  num_minibatches: 32
  gamma: 0.99
  gae_lambda: 0.95
  clip_eps: 0.2
  init_std: 0.2
  learnable_std: false
  ent_coef: 0.0
  vf_coef: 0.5
  max_grad_norm: 0.5
  activation: tanh
  anneal_lr: false
  weight_decay: 0.0
  normalize_env: true
  debug: false
  n_seeds: 1  # will automatically take seeds from 1 to n_seeds
  vmap_across_seeds: true
  validation:
    active: true
    num_steps: 100
    num_envs: 100
    num: 10 # set to 0 to disable validation
    # if none of the following is set, only the mean reward and episode length is logged
    quantities: # quantities to be used for the distance measures below
      - JointPosition
      - JointVelocity
      - RelSitePosition
      - RelSiteVelocity
      - RelSiteOrientation
    measures:
      - EuclideanDistance
      - DynamicTimeWarping
      - DiscreteFrechetDistance
    rel_site_names:
      - upper_body_mimic
      - left_hand_mimic
      - left_foot_mimic
      - right_hand_mimic
      - right_foot_mimic
    joints_to_ignore:
      - pelvis_tx
      - pelvis_tz
      - pelvis_ty
      - pelvis_tilt
      - pelvis_list
      - pelvis_rotation

# Define control subconfigs
control_config:
  position_control:
    control_type: PDControl
    control_params:
      p_gain:
        - 300  # back_bkz_actuator -> torso
        - 100  # l_arm_shy_actuator -> shoulder
        - 100  # l_arm_shx_actuator -> shoulder
        - 100  # l_arm_shz_actuator -> shoulder
        - 100  # left_elbow_actuator -> elbow
        - 100  # r_arm_shy_actuator -> shoulder
        - 100  # r_arm_shx_actuator -> shoulder
        - 100  # r_arm_shz_actuator -> shoulder
        - 100  # right_elbow_actuator -> elbow
        - 200  # hip_flexion_r_actuator -> hip_pitch
        - 200  # hip_adduction_r_actuator -> hip_roll
        - 200  # hip_rotation_r_actuator -> hip_yaw
        - 300  # knee_angle_r_actuator -> knee
        - 40   # ankle_angle_r_actuator -> ankle
        - 200  # hip_flexion_l_actuator -> hip_pitch
        - 200  # hip_adduction_l_actuator -> hip_roll
        - 200  # hip_rotation_l_actuator -> hip_yaw
        - 300  # knee_angle_l_actuator -> knee
        - 40   # ankle_angle_l_actuator -> ankle
      d_gain:
        - 6  # back_bkz_actuator -> torso
        - 2  # l_arm_shy_actuator -> shoulder
        - 2  # l_arm_shx_actuator -> shoulder
        - 2  # l_arm_shz_actuator -> shoulder
        - 2  # left_elbow_actuator -> elbow
        - 2  # r_arm_shy_actuator -> shoulder
        - 2  # r_arm_shx_actuator -> shoulder
        - 2  # r_arm_shz_actuator -> shoulder
        - 2  # right_elbow_actuator -> elbow
        - 5  # hip_flexion_r_actuator -> hip_pitch
        - 5  # hip_adduction_r_actuator -> hip_roll
        - 5  # hip_rotation_r_actuator -> hip_yaw
        - 6  # knee_angle_r_actuator -> knee
        - 2  # ankle_angle_r_actuator -> ankle
        - 5  # hip_flexion_l_actuator -> hip_pitch
        - 5  # hip_adduction_l_actuator -> hip_roll
        - 5  # hip_rotation_l_actuator -> hip_yaw
        - 6  # knee_angle_l_actuator -> knee
        - 2  # ankle_angle_l_actuator -> ankle
```

Has anyone else encountered this behavior? Could it be an issue with the reward config, initial pose, or control instability? Any suggestions would be greatly appreciated!

Thanks in advance!

boris-il-forte, 2025-07-22:
Most likely is a JAX issue. This happens due to NaNs. Try to downgrade to JAX 0.5.2

ShiweiLiu2002, 2025-07-22:
@boris-il-forte Thank you so much for your reply! It does seem like the issue was indeed caused by JAX — your suggestions were extremely helpful!

closed 2025-07-22 by ShiweiLiu2002

minolta001, 2025-07-23:
I tried to train H1v2 with Deepmimic but got the similar result of immediate ending. I have downgrade jax to 0.5.2 but it still doesn't work. May I have any suggestion from you? @boris-il-forte

minolta001, 2025-07-23:
@ShiweiLiu2002 Thanks so much! The 0.5.2 jax I installed still has some issue, but the jax you specified works!

Meanwhile, may i know, after you change the control type to PD control, how's the performance of H1? I trained H1v2 with PD controller, but the performance is way worse than H1 trained with default controller.